FreddyVRetro/ISA-PicoMEM #88: MTCP netdrive fails to connect

raidersan, 2025-01-21:
It looks like NETDRIVE from the MTCP product stack fails with an immediate timeout when trying to request a file from the server ("netdrive connect x.x.x.x:port d:")
The server does see the request coming through and seems to serve it but the client errors on a immediate timeout.
This was tested with a very good connection to the WIFI (1m away with no case cover on), with ping reporting 1 to 4 ms and no packet loss. Other MTCP utilities seem to work fine

FreddyVRetro, 2025-01-22:
Hi,

Can you tell more if there is an error message ?

mbbrutman, 2025-01-22:
I believe I've already been working on this problem with this person.  If that is the case, please, let's stick to the email thread we already have going and not draw more people in until it is needed.

FreddyVRetro, 2025-01-24:
Hi,

@mbbrutman found the bug and I just released a new PM2000 that should fix it.
(Rev 0.4)
Can you please test it ?

closed 2025-02-11 by FreddyVRetro

raidersan, 2025-03-03:
Hi both,

Thanks for looking into this. I can confirm that the problem has been solved and I can now use the most common mTCP programs that were failing, including FTP, ping, netdrive, telnet.
Thanks to both of you for the work and great products

Regards,
JFD