FreddyVRetro/ISA-PicoMEM #89: IBM AT 5170 disk issues

artakaama, 2025-01-24:
I have an IBM AT with the first type mainboard and BIOS (84) and 640k RAM (with an extension card) and the original combo floppy and disk controllers with a 5.25" and a working original disk attached.
PicoMEM (1.5) initializes and is configurable. I made a disk image (22Mb) in the menu and put it on hdd1 so as not to conflict with the physical drive. I can boot from floppy and disk. The sd-card image shows up but fdisk complains it is not accessible, but lets me write a partition table anyways. After reboot, these changes disappear. This behaviour is the same for PC DOS 3.3 and MS DOS 6.22. The image size is just 22Mb.

When disconnecting or removing the physical disk, this behavious does not change. I did not remove the combo disk controller, as I want to be able to backup my physical drives and transfer files from the SD card.

The HD adress range for the disk controller starts at 1F0h.

FreddyVRetro, 2025-01-31:
Hi,

Your Original disk is bootable ?
What is its OS version ?
There are also HDD images on the GitHub you can try

It is not possible to boot on the 2nd disk, and maybe you can't move the physical disk to 2nd drive ?

artakaama, 2025-02-10:
Hello Freddy :)

Yes the physical disk is booting fine, with or without the Pico installed (with hdd0 set to none). It is a type 2 with PC DOS 3.3.
Is it possible the floppy/disk controller interferes with the Pico? I tried moving the adress range on the disk controller, but the pico disk images still were not writeable for fdisk (booting from floppy into MS DOS 6.22).

Do I need a different BIOS? It is the ancient first gen. IBM AT Bios...

FreddyVRetro, 2025-02-11:
Hi,

No idea,
you can also create a disk image using a PC emulator, like PCEM