FreddyVRetro/ISA-PicoMEM #90: PicoMem not showing at boot Tandy 1000 RLX

rbarquer, 2025-02-11:
Hi, I have tried changing the IRQ, but it isn’t working. The PicoMem works well on my Tandy 1000 RL HD but not on the RLX. PicoMem BIOS not showing at boot. says Init: … End Any ideas?

[image]
[image]

FreddyVRetro, 2025-02-11:
Hi,
I think I saw somewhere that Tandy RLX need another BIOS to work on the PicoMEM.

Anyway, maybe try the PREBOOT option in the config.txt file ?

FreddyVRetro, 2025-03-10:
Hi,
Can you confirm me and check so that I can close this issue ?

I updated the Wiki with Tandy 1000 RLX section.

rbarquer, 2025-03-13:
Please close it, with the help of Cloudschatze from forum.vcfed.org I was
able to update the ROM on the Tandy 1000RLX and the issue went away!

FreddyVRetro, 2025-03-13:
Nice, happy for you.
Don't hesitate to test the latest Firmware :)

FreddyV