FreddyVRetro/ISA-PicoMEM #92: Mouse not working on PS/2 and Serial in 486 - Crash when moving mouse in Win3.1

Chrysopidae, 2025-02-19:
Using a Serdaco Edition 1.20a card with the January '25 firmware:

- The mouse does not function when connected to the motherboard's PS/2 port.
- When using a serial connection via a Goldstar GW2760 controller, the mouse is also not recognized.
- Using a HIDman adapter; the keyboard works without issues.
- When plugged into the PS/2 port, touching the mouse in Windows 3.1 causes the system to crash.
- Windows is running in 256-color mode; other modes have not been tested.
- Using the standard CT Mouse driver.
- During boot, the mouse is detected on both PS/2 and Serial connections but does not function in MOUSETST or any other application.
- I don't have a separate bootable option, so can't test without the PicoMEM in this machine.
- Enabling or disabling USB and/or different card IRQ on the PicoMEM makes no difference.

**Hardware:**
- ASUS VL/I-486SV2GX4 motherboard
- Intel 486DX4 processor
- VLB graphics card (QDI Cirrus Logic VGA5426)
- Soundblaster CT1600

FreddyVRetro, 2025-02-22:
Hi,

Is it just when the PicoMEM Board is present or do you run the PMMOUSE Driver as well ?

The PicoMEM Does not decode the PS/2 port :(
Did you try removing the PicoMEM IRQ jumper ?
Disable everything ?

Chrysopidae, 2025-02-23:
Thanks, Freddy! I ran some more detailed tests.

I was actually able to boot directly from my BlueSCSI using the SCSI adapter (which I typically use for the CD-ROM), allowing me to test the behavior without the PicoMem.

When using the motherboard’s PS/2 port, I use the standard CTMouse driver instead of the PMMouse driver. Previously, I also used the PMMouse driver to enable USB mouse support via PicoMem, which works—though not in Windows, as expected.

With the PicoMem installed, even with the jumper removed (IRQ 0) and all options disabled, the system either crashes in Windows or produces random mouse input in DOS when moving.

In Windows 3.11, I receive the following error after moving the mouse:
**Internal Stack Overflow
System Halted**

However, without the PicoMem, when booting from a hard disk image using the SCSI adapter, the mouse is recognized and functions correctly in both DOS and Windows when connected via PS/2.

Interestingly, when the SCSI adapter BIOS is enabled (which is required for booting from SCSI directly), it overrides the PicoMem, allowing the PS/2 mouse to work—even with the PicoMem still inserted.

Additionally, a mouse connected via the serial port did not work in any situation, which might indicate an issue related to either my configuration or the HIDman.

Chrysopidae, 2025-03-08:
Additional test: PS/2 functions correctly with Windows 95, and also serial works when Pico is inserted and enabled (using to boot the Win95 image).

When using the PMMOUSE driver with PS/2 in Windows 3.11, the system freezes immediately upon mouse interaction. With the standard CTmouse driver, a stack overflow error occurs. Serial does not function, though this could be due to a driver and HIDman compatibility issue.

In DOS, mouse usage results in erratic movements and random clicks (e.g., in DOOM), regardless of whether the PMMOUSE driver or a standard driver is used. Other drivers, such as Logitech, produce the same behavior (also stack overflow in Win3.11).

Let me know if you would like me to do any other specific test.

FreddyVRetro, 2025-03-10:
Hi,

Thanks for all thee infos.
With the added infos it looks like a problem in one redirected IRQ.
PicoMEM redirect only the IRQ13 and its hardware IRQ that is different than the PS/2 one.

I am busy on other things, but I promice, I will take a look at that in the future.
This "Bug" may generate other instability/Issue not discovered yet.

B0wm4n, 2025-04-15:
I wanted to start off with I just received my PicoMem V1.14 [Firmware PM_W_3_11_25] and I wanted to say I'm blown away with how easy it's been to add EMS and solid state storage to my machine.  IBM Model 25 (286) 

I too am experiencing the weird Mouse Issues.  Currently I am using the native PS/2 mouse port and Mouse.com (Microsoft V 8.20).  

Similar scenarios, I am able to use the mouse once a game is loaded (SIMCITY/WOLF3D/Civilization) although touching the mouse during the game load will often crash the computer.  It also occurs in DOS, during a DIR command and moving the mouse will crash the computer.  

Windows 2.X works until the mouse is touched.  

Checkit indicates that the Mouse is using IRQ 12

Chrysopidae, 2025-04-27:
Conventional is all system (S).

Video & ROM/EMS distribution looks like this:
xxxxxxVVRR--BB------RRRR
Ignore Video Segment: [On ], rest is Off

Let me know if this is what you need or if you referred to something else. Thanks!

FreddyVRetro, 2025-05-06:
> Conventional is all system (S).
> 
> Video & ROM/EMS distribution looks like this: xxxxxxVVRR--BB------RRRR Ignore Video Segment: [On ], rest is Off
> 
> Let me know if this is what you need or if you referred to something else. Thanks!

Hi,
PC Total RAM amount is in the setup first page, not on the RAM page :)

Chrysopidae, 2025-05-06:
> Hi, PC Total RAM amount is in the setup first page, not on the RAM page :)

Oh there :-). 
System Memory: 639 - Floppy Drives: 1 - Hard Disks: 0
PicoMEM IRQ: 7

B0wm4n, 2025-05-09:
> Hi,
> 
> So I confirm the bug you have is fixed. Will be in the next firmware :)

Looking forward to the next release, thanks!

Chrysopidae, 2025-05-09:
> Hi,
> 
> So I confirm the bug you have is fixed. Will be in the next firmware :)

Excellent, you're the best! Looking forward to the new release then to test :-).

FreddyVRetro, 2025-05-13:
I keep it open so that other can see it is solved.

FreddyVRetro, 2025-06-13:
I suppose I can now close this issue :)

closed 2025-06-13 by FreddyVRetro