FreddyVRetro/ISA-PicoMEM #93: USB LPT adapter on the PicoMEM

themustysmell, 2025-02-24:
Hello.

Is it possible to use an LPT to USB converter on the PicoMEM USB port? If not, would it be technically possible to implement this in future releases?

My requirement would be to use either a ZIP 100 drive or an LPT Ethernet adapter on it.

Many thanks and regards

FreddyVRetro, 2025-03-10:
Hi,

This require a software/Driver that probably does not exist.
It is also extremely rare to have a PC without parallel port.

Regards,
FreddyV

themustysmell, 2025-03-13:
Hello FreddyV. 
My PC (an IBM PS/2 Model 30) has an LPT port on the motherboard. A Xircom LAN adapter is connected there. A second LPT port, ideally via the PicoMEM, would be helpful for me to be able to use an additional ZIP drive.

FreddyVRetro, 2025-05-13:
> Hello FreddyV. My PC (an IBM PS/2 Model 30) has an LPT port on the motherboard. A Xircom LAN adapter is connected there. A second LPT port, ideally via the PicoMEM, would be helpful for me to be able to use an additional ZIP drive.

I understand, but it is not in the plan.
PicoMEM now support direct mapping of a USB key and the MicroSD.
Use of ZIP drive should remain on original hardware.

closed 2025-09-29 by FreddyVRetro