FreddyVRetro/ISA-PicoMEM #94: When using it with Pocket8086, it stops with an IO Port Error.

tiramisu376, 2025-03-06:
When using it with Pocket8086, it stops with an IO Port Error.

I thought it might be the Adlib module, so removed it and tried setting the PRE BOOT specification in config.txt, but it didn't change.

It started up on HAND386.

[image]

tiramisu376, 2025-03-08:
Addition:
The above error is displayed when the CPU is an 8086 or V30, but the IO Port error is not displayed when the CPU is a V20.
However, because the Pocket8086 has limited access to the ROM to F000h to F3FFh, it cannot access the PicoMEM BIOS located at D000h.

FreddyVRetro, 2025-03-10:
Hi,

I have a Book8088, Pocket8086 and Pocket 386.
The Pocket8086 does nto work for me as well.
These machines are "out of spec" for the ISA signals and it is the only machine I know with I/O not working.

I don't know if we can contact the machine creator to inform him.....

The Pocket8086 sample the IOCHRDY signal too early (The signal used to increase the cycle duration)
Then, the I/O is done too fast.
I can still increase a little the IOCHRDY signal speed in the code, but only by some nano second, not sure if it is sufficient....

FreddyVRetro, 2025-05-06:
Hi,

So, the Pocket386 is bugged on its ISA bus, PicoMEM can't work without modifying it. :(

i will check for Pocket8086 later, as I have one as well.

closed 2025-05-06 by FreddyVRetro

rvalles, 2026-11-15:
I confirm Pocket8086 (with V30) at 10MHz is no good with PicoMEM2 2.0.2 / PM2-10-4-26. The boot banner is partially printed, then it's gibberish and crash.

It seems to work fine at 4MHz.