FreddyVRetro/ISA-PicoMEM #95: Card not seen on Micro8088 system.

prime6809, 2025-03-08:
PicoMEM rev 1.14, programmed with latest firmware (PM_W_Jan1.uf2).

Brand new 32GB PNY SD card (previously tried 16GB Kingston).

My config.txt contains : 

#Config for Book8088 rename to config.txt
BIOS C800
PREBOOT

I have a Trident VGA card with it's BIOS at C000-C7FF, UMB RAM at D000-EFFF, and Micro8088 BIOS from F000-FFFF.

IRQ on the board is set to 7.

Power on the system and it boots as if the PicoMEM was not there, though the green LED on the Pico-W does come on, which I believe indicates that it initialized correctly.

Any clues as to what might be wrong?

Cheers.

Phill.

FreddyVRetro, 2025-03-08:
Hi,
There are other post with the same kind of problem here.
The psu 5v is surely too low.
Try to start the picomem by connecting its usb to a PC

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

prime6809, 2025-03-08:
Humm, 

The voltage as measured on the VBUS or VSYS and GND pins on the pico is 4.6V, as measured the other side of the input diode is 5.0V from my own experience interfacing Pi Picos to 5V logic this should be high enough.

Cheers.

Phill.

FreddyVRetro, 2025-03-08:
Can you test with the usb cable?

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

prime6809, 2025-03-08:
Right progress, disabling the UMB at d000 on the Micro8088 card has allowed me to see the PicoMEM Init message, suggesting it's ignoring the contents of config.txt.

Cheers,

Phill.

FreddyVRetro, 2025-03-08:
Not ignored,

the pc scan the option rom too early.

Bios c800 must be the first line or remove it

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

prime6809, 2025-03-09:
Humm seems to not be reading the SD card, I can now get into the BIOS fine, but on the disk page I get an error about "SD card not present or error". Also on the EMS page it claims that the PSRAM is not present.

I've tried a couple of cards but I have the same problem with both.

Cheers.

Phill.

FreddyVRetro, 2025-03-10:
Hi,
If both uSD and PSRAM fail, this is surely a weak soldering on one of the 3 pin connecting the pico to the PSRAM and uSD (They are in common)
Pin 7 and 9 10 from the top right.
You can see the trace going from these pin to the PSRAM chip.
Generally, we can see it is nto soldered correctly by eyes, or zoom on it with a phone.