FreddyVRetro/ISA-PicoMEM #98: Eagle Spirit XL ignoring S keypress

McJones9631, 2025-03-30:
I have a Eagle Spirit XL that I installed my PicoMEM into. The SD card is as follows:

/FLOPPY/
- (empty)
/HDD/
- D33_EN.IMG
config.txt

My config.txt is as follows:

PREBOOT

HDD0: D33_EN.IMG

The green light on the PicoMEM does turn on, but holding S upon boot just kicks me back to the Eagle's "No boot sector found". The Pico is in the 2nd ISA slot, between the floppy controller and the CGA card, and is jumpered at IRQ5.

I apologize if this is a simple solution/negligence on my end, it is my first time playing with a card like this.

FreddyVRetro, 2025-04-13:
Hi,

Use PREBOOT is taking the risk that the PC BIOS did not initialize the Keyboard yet

Why do you need to add Preboot ?

FreddyVRetro, 2025-05-06:
Hi,

The S Key id the key in qwerty layout.

Is the S located in the correct location on your keyboard ?

McJones9631, 2025-05-06:
Apologies for the late response.

So the Eagle _does_ boot to the PicoMEM, but only after I force it to run through a memory test by holding down T at boot. If I do not do this step, the machine does not boot to anything. After the memory test, I am able to hold down S and the PicoMEM BIOS pops up just fine, and exiting that my HDD image works perfectly.

Mellvik, 2025-05-15:
FWIW - using PREBOOT apparently does this, prevents the keyboard from working, on many machines - such as my Compaq Portable Plus. Removing PREBOOT (which was experimental anyway in my case) makes it work.