FreddyVRetro/ISA-PicoMEM #100: Set  $C800-$CFFF address?

andreacampanella, 2025-04-24:
I have this motherboaed : 
https://theretroweb.com/motherboards/s/a-trend-4gpv3

someone wrote about it:
Cannot cache the ROM BIOS from $D000-EFFF. Only caches ROM BIOS from $C800-$CFFF. (It can cache the VGA BIOS and system BIOS.) This must be a chipset problem as I tried several BIOSs for this Chipset, and they either hide caching entirely or only allow you to cache $C800-CFFF.

could be possible to set a different address? if so, how can I do it without a working computer first?

electricbolt, 2025-04-25:
https://github.com/FreddyVRetro/ISA-PicoMEM/wiki/Configuration-files

FreddyVRetro, 2025-04-27:
Hi,
Your question is answered ? Can I close the case ?

closed 2025-05-06 by FreddyVRetro