FreddyVRetro/ISA-PicoMEM #102: Tries my PicoMem card in my Schneider PC 1512 but floppy boots only when card is removed

lollisoft, 2025-05-12:
Hi,

I have a PicoMem card and there is a SD card I have temporarly borrowed from my raspi that has a 256 MB Fat Boot partition. It is a Fat32 I think.

But that seems not to be the issue. My issue is hat if I insert the card, the original floppy drive did not boot from the same disk that boots when the card is removed.

I have let IRQ settings to factory IRQ 7. The original hard drive (HD20 model) has been deactivated and the card has been removed for that test.

Do you have any ideas?

Is there a BIOS hotkey to look into?

Thanks, Lothar

FreddyVRetro, 2025-05-13:
Hi,

What floppy do you want to boot ?

The PicoMEM does not support DMA from the emulated RAM, so if you enable the RAM emulation, Floppy boot or application using the emulated RAM do fo Floppy data transfer will fail.

Like early version of DOS.

Next gen PicoMEM (Not released) support DMA from emulated RAM

lollisoft, 2025-05-13:
I tried to boot from hardware floppy drive. The original.

After reading the wiki, I also had to update the firmware and then initialized the SD card as described. Using the Book 8088 settings and configured the 200 MB DOS 3.3 image. I tried no IRQ, IRQ 7 and 5. The original MFM controller had a setting on IRQ 5 as I was
able to read old spec pdf.

Next I try the settings for the PPC512 and then on a real PPC512 as a fallback to see the card workin in a known to be working setup.

lollisoft, 2025-05-14:
Hey,

I got it working. According a post here https://github.com/FreddyVRetro/ISA-PicoMEM/issues/95 I have also tried to power from an external power source. In my case a power bank.

Is it really needed to power externally with a Schneider PC 1512 or are there some power lines not connected on that board? I know that another user got the PC1512 working with that card, but not how.

Neonetheless I can go a step further :-)

FreddyVRetro, 2025-05-16:
Hi,

Amstrad/Schneider 1512 is the reason why there is the "U10" chip on the card to re generate a "Power good" signal.
It has a noisy PSU

FreddyVRetro, 2025-06-13:
> Hey,
>
> I got it working. According a post here [#95](https://github.com/FreddyVRetro/ISA-PicoMEM/issues/95) I have also tried to power from an external power source. In my case a power bank.
>
> Is it [...]

Hi,
did you do progress ?