FreddyVRetro/ISA-PicoMEM #103: NE2000 emulator emitting incomplete MAC address

Mellvik, 2025-05-15:
Compaq Portable Plus running TLVC Linux: The ne2k driver (which works well with all known variants of the NE2k/ne1k and compatibles) finds the interface OK, but gets an incomplete  MAC address. At power-on the last 3 bytes of the MAC address are 0 (the first 3 are correct). Reboot, and the 4 last bytes are zero. Reboot again and only the first byte of the MAC address is non-zero (and correct by the way).

After that, it stays that way.

Whether the emulation works in this setting is still unknown for obvious reasons, but incoming ARPs end up in the ARP table, which is promising.

```
ttyS0 at 0x3f8, irq 4 is an 8250
30 ext buffers (base @ 0x7880), 8K L1-cache, 20 req hdrs
eth: ne0 at 0x300, irq 3, (ne2k) MAC 28:c1:f7:00:00:00, flags 0x80, bufs 2r/0t
eth: wd0 at 0x240, irq 2, ram 0xc800 not found
xd: MFM disk controller @ 0x320, irq 5, DMA 3
xda: CHS 306/4/17/128 (preset)
df: Floppy controller, FDC 8272A @ irq 6, DMA 2
df0, df1 set to 360k/PC (type 1)
Floppy cache 5k, available 5k
...

```

Mellvik, 2025-05-15:
Some more insight into this problem (and a solution).

Looking under the hood it turns out that what the PI is returning to the driver (the first 32 bytes) is the following:
```
28 cd c1 12 f7 41 00 00 00 00 00 00 57 57 57 57 57 57 57 57 57 57 57 57 57 57 57 57 57 57 57 57
```

The full Mac address is there. Which is great - except it is delivered in the WRONG bytes. A ne2k interface is supposed to return each of the 6 elements of its MAC address in the low part of the first 6 words of the 'station PROM'. Further - the '0x57' in the upper part is wrong.  Here's a summary of what a driver would expect from a NE2k NIC:

```
void INITPROC ne2k_drv_init(void)
{
        int i, j, k;
        word_t prom[16];/* PROM containing HW MAC address and more
                         * (aka SAPROM, Station Address PROM).
                         * PROM size is 16 words, the low byte holding the info
                         * (newer cards may have proprietary info in the high byte).
                         *
                         * QEMU duplicates the low byte to the high byte, which we use
                         * to set the QEMU flag (for ftp). This feature is currently
                         * unused as there is no way to pass that info up to ftp.
                         *
                         * The MAC address occupies the first 6 words, followed by a 'card signature'
                         * which is usually empty except for bytes 28 and 30, which contain
                         * a 'B' if it's an 8bit card, 'W' if it's a 16bit card.
                         * If a NIC doesn't implement this feature, 16-bit mode
                         * may be forced using the flag field in /bootopts - just like a 16bit
                         * card may be forced to run in 8bit mode.
                         */
        byte_t *cprom, *mac_addr;
```

The more or less obvious reason for the address being placed like that was that at the point when this info is being read by a driver, there is no way for the system to know whether it's an 8 or 16 bit card, and for the NIC to know whether it's talking to an 8 or 16 bit bus. So keeping the MAC address in the low part of every word is safe. It always works. 

The PicoMEM is an 8but card, so bytes 28 and 30 should be 0x42.

Hope this helps. And FWIW, when manually setting the MAC address, network traffic runs fine on my system. Slower than a real NIC but fine.

FreddyVRetro, 2025-05-15:
Hi,

Do you have the link to the driver source ?

Mellvik, 2025-05-16:
[tlvc/arch/i86/drivers/net
](https://github.com/Mellvik/TLVC/tree/master/tlvc/arch/i86/drivers/net)

Check out this directory - there is a C part and an ASM part. Let me know if you have questions. I rewrote this driver for ELKS (then TLVC) some years back.

Mellvik, 2025-05-16:
Here is a related (networking that is) issue: I've been switching between two SD cards, one 4GB, one 16GB. Both cards are found and reported OK by the PicoMEM POST screen, the content is found and acted upon by the Pi (like, the POST screen reporting 'connected' to the WiFi network and the PICOMEM file gets written). Both cards.

However, when using the 4GB card, the ne2k interface is never found. When using the 16GB card, the ne2k is always found. Apparently the driver probe gets different results depending on which SD card is used. Weird. If I hadn't had the 16G card around - and switched between them for convenience, I would have easily concluded that the PicoMEM network emulator just does not work...

FreddyVRetro, 2025-05-16:
Hi,

it was probably due to a corruption in the picomem.cfg file, you can try to delete it.

FreddyVRetro, 2025-07-12:
Hi,

For the detection problem on the TLV Linux, I have no idea how to correct the code with what you provided.
I did not write the ne2000 emulation code

For the difference between 4GB and 16GB SD, is it still with TLV Linux ?

Mellvik, 2025-07-13:
FWIW, the fix goes like this - 
Old code in `ne2000.c` line 350:

```
        ne2000->macaddr[0] = ne2000->physaddr[0];
        ne2000->macaddr[1] = ne2000->physaddr[1];
        ne2000->macaddr[2] = ne2000->physaddr[2];
        ne2000->macaddr[3] = ne2000->physaddr[3];
        ne2000->macaddr[4] = ne2000->physaddr[4];
        ne2000->macaddr[5] = ne2000->physaddr[5];
        // ne2k signature
        for (i = 12; i < 32; i++) {
                ne2000->macaddr[i] = 0x57;
        }  
```
correct code:
```
        ne2000->macaddr[0] = ne2000->physaddr[0];
        ne2000->macaddr[1] = ne2000->physaddr[0];
        ne2000->macaddr[2] = ne2000->physaddr[1];
        ne2000->macaddr[3] = ne2000->physaddr[1];
        ne2000->macaddr[4] = ne2000->physaddr[2];
        ne2000->macaddr[5] = ne2000->physaddr[2];
        ne2000->macaddr[6] = ne2000->physaddr[3];
        ne2000->macaddr[7] = ne2000->physaddr[3];
        ne2000->macaddr[8] = ne2000->physaddr[4];
        ne2000->macaddr[9] = ne2000->physaddr[4];
        ne2000->macaddr[10] = ne2000->physaddr[5];
        ne2000->macaddr[11] = ne2000->physaddr[5];
        // type of interface: 8bit (ne1k)
        ne2000->macaddr[28] = 0x42; 
        ne2000->macaddr[30] = 0x42; 
```
... or just skip the odd bytes of `macaddr`. Duplicating them like this is what QEMU does for good measure. I haven't seen any real NIC cards doing that, but it doesn't hurt.

Just as important as the address is to get rid of the 0x57 which is added to the rest of the 32 byte block. 0x57 is ascii 'W' indicating a WORD size (16bit) interface, I.e. a ne2k. This is a ne1k (8bit) interface emulation and the correct value (in bytes 28 and 30) is 0x42 (ascii 'B' for BYTE). The rest of the bytes in the 12-32 range can be anything ('signature block' is about right). Putting like 'picoMem' in there (say from 20 to 26) would make it possible for the host to identify this particular interface for whatever that might be worth.