FreddyVRetro/ISA-PicoMEM #108: Cannot select HDD images in BIOS

p-rintz, 2025-07-12:
Hi there,

I bought a picomem form serdaco and it seems to generally be working fine on my DECpc 450d2lp.

However I am unable to select HDD images that either were created via the BIOS or via Winimages.

They just don’t show up in picomem bios.

I created the first three images via the bios and the last via Winimage on Windows 98.
None of them are selectable (or show up at all) in the bios.

[image]

Thanks for any help.

FreddyVRetro, 2025-07-12:
Hi, Create image from BIOS does not work anymore in the last version, I need to debug it

Can you delete the first 3 ones to see ?
Also try another SD .

p-rintz, 2025-07-12:
Tried both things and it didn’t help. Still can’t select any HDD images.

I just found out I also have an issue with the ctrl+shift+f2 switching of floppy images. 
They just don’t… as in the first mounting works fine, afterwards it just doesn’t change.

I’ll try downgrading the firmware version.

p-rintz, 2025-07-13:
I was able to flash an old firmware with some difficulty.

And it seems that in addition to the inbuilt tool being broken I somehow also created a non-working image with WinImage.

This is now working with the old firmware created images.

Ill open a different issue for the Floppy changing problem.

closed 2025-07-13 by p-rintz