FreddyVRetro/ISA-PicoMEM #111: v0.8 Can't create HDD images

uniQ168, 2025-08-05:
Hi,

i just installed my new PicoMEM board in a PC10-III and tried to create a new hdd image file.
It created the image but I can't select it in the BIOS. I checked the image file on the sd card and it's just 0kb in size.

I'm on the latest firmware V0.8 / 2025-05-24.

uniQ168, 2025-08-05:
Downgraded to V0.7 and it works.

FreddyVRetro, 2025-08-05:
> Downgraded to V0.7 and it works.

Hi,
The new test firmware just released fix it :)

andypants152, 2025-08-14:
with the new test firmware i was able to make some HDD images, but on my Tandy 1000 TL with a 1.14 model; it is now unable to access the hdd images mounted outside the 622 image or default deskmate rom...

andypants152, 2025-09-25:
> Boot Does not work at all ?
> Did you add TANDY in the configuration file ?
> (config.txt on the SD root directory)

I didn't set the config file but the picomem bios picked up where the rom
was located. I downgraded and was able to access the images again, just
can't create new ones on the picomem.

not a huge issue: since i can just pop out the SD card on the earlier
version.

andypants152, 2025-11-08:
> Hi,
>
> Did you try the latest TEST firmware (August 2025) ?
> And with no RAM emulated ?
>
> Also, Set TANDY in the config file is normally mandatory, I don't
> understand how an older firmware work without that...

With the PicoMem Boot Code enabled it boots to the Tandy DOS 3.3/Deskmate
on ROM

I don't see any mention of the Tandy config file setting in the docs?