FreddyVRetro/ISA-PicoMEM #112: Unable to access BIOS menu

0xbretwashere0x, 2025-08-07:
Hello. I have a PS/1 2121, and I’m having issues using my picoMEM. I updated the firmware to the four latest firmware, and now when I brought the computer, I get the following message:
PicoMEM BIOS (2025-03-11) Init: …. End

I am not able to access the set up menu from its option room. I’ve tried the following configurations in the config.TXT file:
PREBOOT
BIOS C800

I have also tried removing PREBOOT and the BIOS address, and have set the BIOS address to both C800 and D000. I have also sent the IRQ to IRQ 3, 5, and 7. 

Can someone please be of assistance with this issue. Even replacing the firm or back to its original (I don’t recall what the date of the firmware it was) does not solve the issue. Thank you.

0xbretwashere0x, 2025-08-07:
I wanted to add that the only firmware that works and allows me to access the menu is the 2024-07-20 17:22:36 build. I have BIOS C800 and PREBOOT in the configuration file.

FreddyVRetro, 2025-10-01:
Hi,

Maybe also try to remove the picomem.cfg file

0xbretwashere0x, 2025-11-16:
Sorry for the delay. I will attempt removing the cfg on the latest firmware and will report back