FreddyVRetro/ISA-PicoMEM #117: Help for a user who lacks reading comprehension

splaspood, 2025-09-04:
I just recently received my first PicoMEM from TexElec, and it's brought life to my 80286!

At first I couldn't get any floppy images to work, and I couldn't figure out why.   I thought maybe I had a bad firmware revision (I didn't) or some other issue that might've been caused by the rest of the system hardware which was of an unknown quality at that point.

Eventually, after re-reading README.md in images/ over and over I caught that floppy images should be in a FLOPPY/ directory off the root of the micro sd...   I had simply copied the contents, recursively, of images/ into the root of my microsd card.    Which would have worked fine, except the floppy images are in a directory named FDD in git, not FLOPPY.

Obviously this is user error, and it's clearly documented correctly, but I wonder if the FDD directory should be renamed to FLOPPY for silly users like myself.

FreddyVRetro, 2025-09-04:
Hi,

Sorry for that :(

I don't understand, where it is written that Floppy folder is FDD ?

splaspood, 2025-09-04:
It's not actually written anywhere in the documentation, but within the git repo, the directory is named FDD rather than FLOPPY so you can't just copy the stuff from get onto the microSD you need to rename the directory.   Since I didn't read the documentation close enough, it took me some time before I realized that was my problem.

splaspood, 2025-09-08:
I see you ended up renaming the directory within the repo.    Future people who didn't RTFM (like me!) thank you. :).    

(On an almost completely unrelated note, my picomem is working great!   Thanks for producing it.)

closed 2025-09-08 by splaspood