FreddyVRetro/ISA-PicoMEM #123: ROM Checksum error at CE00

RetroBard, 2025-09-28:
I haven't been able to get the PicoMem boot menu to appear on my Schneider EuroPC1. I've tried different FR firmwares, using and not using the  PREBOOT option and having BIOS D000 and C800 in the config. The menu hasn't appeared and debug hasn't shown anything PicoMem related in those addresses.

Now, I decided flash the latest FR firmware PM_W_3_11_25_FR, and based on my research on the Schneider and option ROM addresses, I had set the BIOS address to CE00.

Now, when I tried booting, I got two errors at bootup:
ROM checksum error at segment CA00
ROM checksum error at segment CE00

I haven't seen those before, so I ran debug as instructed in the FAQ, and now I could see the PicoMem BIOS firmware version text.

So, I'm assuming that the EuroPC1 is actually detecting the PicoMem firmware using that memory address, but it's now failing with a ROM checksum error. 

I hope this information is of some help.

FreddyVRetro, 2025-09-29:
Hi,

Glitch on the PicoMEM may be possible due to power, If you try to power the PicoMEM externally via USB, it may be better, like described here:
https://blog.jakobs.systems/en/blog/20241206-picomem-retro-board/

FreddyVRetro, 2025-12-16:
Hi,

New firmware is released, can you please test with it ?

Regards,
FreddyV

RetroBard, 2025-12-23:
If you mean the November 16 firmware, I’ve already tried that. Still no joy.

RetroBard, 2026-10-09:
[image]
Finally, I’ve managed to get the PicoMEM to be recognized by the EuroPC! It was a matter of finding a correct memory address, CC00 in this case. As you can see, I’m still getting an error, but I’m hoping that it’s because I’m running a card made for PicoMEM 1 in a PicoMEM 2. I need to get another SD card and create it from scratch.

RetroBard, 2026-10-09:
[image]
I put the SD card back into PicoMEM 1, and got a slightly different error, hopefully that might give some indication what’s wrong. 

Both PicoMEMs have the latest firmware. I tried PREBOOT also to force it into settings, but still got the error.

RetroBard, 2026-11-12:
After getting another EuroPC for loan and being able to compare, I was able to determine that I had a broken trace on the ISA bus address line that is related to C800 (SA14). After fixing the line both PicoMem 1 and 2 began working. 
Thank you for the assistance in trying to get this sorted out! Closing the issue.

closed 2026-11-12 by RetroBard